bblanchon/ArduinoJson #2169: Arduino Json 7.4 and boot loop, troubleshooter does not help.

sblantipodi, 2025-04-10:
**Describe the issue**  

I have this code:

```
#ifndef GLOW_WORM_LUCIFERIN_WEBSETTINGS_H
#define GLOW_WORM_LUCIFERIN_WEBSETTINGS_H

const char settingsPage[] PROGMEM = R"=====(<html></html>)=====";

#endif //GLOW_WORM_LUCIFERIN_WEBSETTINGS_H
```

It worked well before Arduino Json 7.4.
Starting from 7.4 this code makes my ESP32 to boot loop.

I changed it this way as per troubleshooter suggestion:
```
#ifndef GLOW_WORM_LUCIFERIN_WEBSETTINGS_H
#define GLOW_WORM_LUCIFERIN_WEBSETTINGS_H

const char settingsPageData[] PROGMEM = R"=====(<html></html>)=====";
const __FlashStringHelper* settingsPage = (const __FlashStringHelper*)settingsPageData;

#endif //GLOW_WORM_LUCIFERIN_WEBSETTINGS_H
```
but this code does not compile:  multiple definition of `settingsPage';
why it says multiple definitions if I used ifndef?

I also tried the other suggestion 
```
#ifndef GLOW_WORM_LUCIFERIN_WEBSETTINGS_H
#define GLOW_WORM_LUCIFERIN_WEBSETTINGS_H

const auto settingsPage = F("<html></html>");

#endif //GLOW_WORM_LUCIFERIN_WEBSETTINGS_H
```

but this one still bootloop.

The weird thing is that the settingsPage object is not even used by Arduino Json but it's used by the WebServer.h library.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v7/troubleshooter/):  
[Paste the report here]

**Environment**  
Here is the environment that I'm using':
Microconroller: ESP32_S3
Core/runtime: espressif32@6.10.0 from PlatformIO
IDE: CLION

```c++
JsonDocument doc;

DeserializationError error = deserializeJson(doc, "{\"hello\":\"world\"}");

// insert code here
```

**Program output**  
If relevant, include the program output.

Expected output:

```
[insert expected output here]
```

Actual output:

```
[insert actual output here]
```

bblanchon, 2025-04-11:
Hi @sblantipodi,

It's absolutely possible that ArduinoJson 7.4 contains a bug, but it's also possible that the upgrade reveals a latent bug in your project.

I cannot make any diagnosis without an [MCVE](https://www.stackoverflow.com/help/mcve/), so please provide one.

If you're unable to create an MCVE, it can mean that there is indeed an undefined behavior somewhere else in your project. If that's your case, try commenting out unrelated pieces of code until the problem disappears.

Best regards,
Benoit

bblanchon, 2025-04-11:
See also #2170.

sblantipodi, 2025-04-11:
hi @bblanchon thanks for the kind answer, I appreciate it.
if I revert Arduino Json to 7.3.x the code does not boot loop.
If upgrade Arduino Json to 7.4 and remove PROGMEM from my code, the code does not boot loop.

It's something related to PROGMEM but PROGMEM in my code is used only by the webserver.h library, it's weird that Arduino Json triggers this behaviour.

I'll try to create an MCVE but it will take time. 
Thanks for your work here.

bblanchon, 2025-04-11:
This very much sounds like #2170.
I already made a patch, and I'll soon publish a new revision.

bblanchon, 2025-04-11:
@sblantipodi, please try with ArduinoJson 7.4.1

sblantipodi, 2025-04-11:
@bblanchon 
this fixed my issue completely.

thank you for such a wonderful lib and for your support Benoît, much appreciated.

closed 2025-04-11 by sblantipodi