bblanchon/ArduinoJson #2170: Ver 7.4 tiny string and ESP32 Arduino Flash String Helper

aaron-neal, 2025-04-10:
I am updating a project that has been happily on 7.1.0 for a long time now.

Instantly, I was experiencing crashes. After connecting up the debugger I was seeing asserts related to the TinyString  work.

In particular this line on the stack:
```_object[F("uid")] = uid;```

After a little playing around, I have concluded there seems to be an issue with the Flashstring helper macro, F()

changing to ```_object["uid"] = uid;``` solved the issue.

Whether I should continue using the F() macro in my projects I don't know,  but for the time being everything 3 chars of less has had them removed and I can continue my dev work.

Many thanks for the library.

bblanchon, 2025-04-11:
Hi @aaron-neal,

Thank you very much for reporting this issue!
I just released ArduinoJson 7.4.1 to address it.
Let me know if it works.

Best regards,
Benoit

closed 2025-04-11 by bblanchon

aaron-neal, 2025-04-11:
@bblanchon, thank you for the rapid response, yes my issues seems to be resolved now.