bblanchon/ArduinoJson #2178: Inspect Defect on Preprocessor.hpp

FrankVanris2, 2025-05-17:
So this isn't a related compile or runtime issue, but I was wondering when I use the inspect tool for my project. Why do I get this error twice? It's currently on the list of high priority:

CWE-0:failed to expand 'ARDUINOJSON_BEGIN_PUBLIC_NAMESPACE', Invalid ## usage when expanding 'ARDUINOJSON_CONCAT_': Unexpected token ')'

It specifies that the issue occurs  in the preprocessor.hpp file. I don't really understand this error however, I think this is more related to the library itself than to my project. My project still compiles properly and runs properly, but the inspect tool tells me that there is some high priority issues related to this library.

[image]

[image]

Preprocessor.hpp:

[image]

bblanchon, 2025-05-18:
Hi @FrankVanris2,

This problem was already reported in #2120.
As far as I can tell, it's a false positive in cppcheck.
We can find many mentions of this error message on the Internet, but I could not pinpoint the actual ticket tracking this issue in the cppcheck project.

Best regards,
Benoit

closed 2025-05-18 by bblanchon

FrankVanris2, 2025-05-20:
ahhh ok gotcha, thanks for letting me know. Was kinda worried if this would cause a bigger issue in the future.