bblanchon/ArduinoJson #2180: is<JsonArray>() not Recognizing the type within a deep object Merge

johnnytolengo, 2025-05-22:
ARDUINOJSON_VERSION "7.2.0"

My testing sandbox: https://wandbox.org/permlink/vQUxzh1zDxib5J0q

I need a deep merge function keeping the old configuration from the previous Json object.
I that function I need to know the type of the object in order to know how to process it, but is<JsonArray>() is returning always Zero.

[image]

Probably is there a better way to do that deep merge keeping the old configuration but for now it's what I have.

Thanks.

bblanchon, 2025-05-23:
Hi @johnnytolengo,

`JsonVariantConst::is<JsonArray>()` always returns `false` because you cannot get a read-write reference from a read-only one.
You must use `is<JsonArrayConst>()` instead.

See [`JsonVariantConst::is<T>()`](https://arduinojson.org/v7/api/jsonvariantconst/is/).

Best regards,
Benoit

johnnytolengo, 2025-05-23:
@bblanchon  ohh I didn't know that,  I applied that "JsonArrayConst" and it worked.

Thank you!
Regards

closed 2025-05-23 by johnnytolengo