bblanchon/ArduinoJson #2183: Version 7.4.1, serialisation of char includes only first letter

BenTubbing, 2025-06-15:
**Description**
I copy / paste the serialisation example - in an empty sketch. 
Just change monitor baud rate to 115200, and run.
The serial output is:
{"s":"g","t":1351824120,"d":[48.75608,2.302038]}
{
  "s": "g",
  "t": 1351824120,
  "d": [
    48.75608,
    2.302038
  ]
}
That is, all char data is clipped after first char.

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at run time
3. The issue concerns serialization
4. Output is incomplete
5. `JsonDocument::overflowed()` returns `false`
6. The empty/missing value is neither an array, object, nor string
7. The output is neither a `char*`, `char[N]`, `MqttClient`, nor `PubSubClient`

**Environment**
* Microcontroller: Arduino DUE
* Core/Framework: Don't know, core for DUE
* IDE: IDE 2, most recent as of today 15/06/25

**Reproduction code**
```c++
The code is copy paste from the website example.
```

**Remarks**
I have used an earlier version of ArduinoJson successfully in the past. So I did not expect this issue. I haven't tried to understand or fix it. But I guess it is useful for you to be informed.

BenTubbing, 2025-06-15:
ADDITIONAL INFO: I just ran the same code on an Arduino MKR 1010 WiFi. This does NOT show the problem.

bblanchon, 2025-06-16:
Hi @BenTubbing,

Thank you very much for reporting this issue.
It has already been reported in #2181, but thanks to you, we now know it only occurs with Arduino Due.

@BenTubbing, could you pinpoint the exact version of ArduinoJson where this problem first occurred?

I'll close this issue as a duplicate of #2182; let's **continue the conversation there**.

Best regards,
Benoit

closed 2025-06-16 by bblanchon