bblanchon/ArduinoJson #2192: Deprecation warning for StaticJsonDocument is harmful on embedded targets

hamidmoradnia, 2025-08-21:
Hi ArduinoJson team,

The new deprecation warning urging users to replace StaticJsonDocument with JsonDocument is creating real problems on long-running embedded systems.

On MCUs with tight RAM and no MMU, variable-size payloads inevitably cause heap fragmentation when memory is acquired/released dynamically. Over days/weeks, this leads to allocation failures and instability. A preallocated, fixed-footprint buffer with StaticJsonDocument avoids that churn: we pay RAM once, then repeatedly clear() and reuse it safely.

Requests:

Please reconsider deprecating StaticJsonDocument, or provide an official way to silence the warning (e.g., a compile-time flag).

If the long-term direction is JsonDocument, please document a fully static, no-heap pattern (with examples) so we can keep a fixed memory footprint without fragmentation.

Clarify best practices for long-uptime devices processing variably sized JSON (streaming, bounded capacity, reuse) without relying on dynamic allocations.

StaticJsonDocument is heavily used precisely because it’s robust on constrained devices. Removing or discouraging it pushes developers toward patterns that degrade reliability in production.

Thanks for considering the embedded use case.

hamidmoradnia, 2025-08-24:
Dear Benoit,

Thank you so much for the detailed and honest answer. Your hard work on the ArduinoJson library is truly appreciated by the community. It has become an essential tool for many of our projects, and we're grateful for all the effort you've put into making it so powerful and robust.

I have a few follow-up questions to better understand my options for long-term projects:

To avoid the memory issues I've described, are you suggesting that my only option is to switch my projects to ArduinoJson version 6?

If I stay with version 7, will my current code (which uses StaticJsonDocument) simply compile and run like version 6, with all the same benefits regarding static memory allocation? Or is it now just a placeholder for the new dynamic allocation method, which will eventually lead to heap fragmentation?

If it's just a placeholder, does that mean I lose the key benefits of version 6's static memory model? Can I still use other new features from version 7 if I stick to the version 6 memory allocation style?

Finally, would it be acceptable to suppress the deprecation warnings in my code to avoid clutter, since I plan to continue using the StaticJsonDocument class for its reliability on constrained devices?

Thank you again for your time and for creating such an incredible library.

Best regards,
Hamid

bblanchon, 2025-08-26:
Hi Hamid,

I'm sorry if it wasn't clear in my previous message, but there is no static allocation in ArduinoJson 7.
As I said, `StaticJsonDocument` is simply a stub on top of `JsonDocument`.
There is, therefore, no benefit to using `StaticJsonDocument` with this version.

If you need static allocation, however, you must switch back to ArduinoJson 6.

Best regards,
Benoit